shastro/NMTLunabotics2023 #29: Robot SubNet

shastro, 2023-04-29:
Need to finish the groundwork for setting up a subnet on the Pi to Jetson.

Currently:
- Both use `dhcpcd` to set static ips on eth0
-- Jetson: 10.0.0.2
-- Pi: 10.0.0.100

The jetson should already be configured to allow ip forwarding.

I set up some routing tables on the pi with the jetson as a gateway

The remaining stuff:
- Need to set up NAT on the jetson
-- I tried to do this but it didn't quite work.

shastro, 2023-05-11:
@AlexBethel Add your updates to this pls

AlexBethel, 2023-05-11:
The Pi had `static routers=10.0.0.1` in its `/etc/dhcpcd.conf`, which mismatches the Jetson having IP address 10.0.0.2, so I changed the Jetson to 10.0.0.1. The Jetson now has `static ip_address=10.0.0.1` in its `/etc/dhcpcd.conf`, but that doesn't work for some reason and we need to set the IP manually anyway; and we have `10.0.0.100 cameras` in the Jetson's `/etc/hosts`. With that we can do
```
sudo route add -net 10.0.0.0/24 gw 129.138.171.209
```
on a dev machine which then makes `ping cameras` succeed, but `ssh cameras` fail with no response; I suspect NMT-Weblogin may be responsible.

AlexBethel, 2023-05-11:
Also `ssh cameras` worked briefly yesterday, and I'm not sure what changed.

AlexBethel, 2023-05-11:
The Jetson is now configured to set its ethernet address via a systemd service, which *might* work better; can't test this right now for logistical reasons of rebooting the Jetson while others are doing work. Also I've determined that the camera pi normally tries to connect to NMT-Weblogin, and if it's done that then it sometimes tries to route TCP response packets over that network, which it's not automatically authenticating to, so ITC blocks the packets. Disabling wlan0 on the pi solves the issue, which I'll do via the `config.txt`.

AlexBethel, 2023-05-15:
Everything works so long as the connecting machine uses `docker_scripts/setup_routes` to initialize routing tables, and has `10.0.0.100 cameras` in their `/etc/hosts`.

closed 2023-05-15 by AlexBethel